Desuuuu/DGUS-reloaded-Klipper #1: Move axes doesnt work

ffnull, 2020-09-19:
Hello. Installs your mod. All fine exept moving axes. Icon doesnt toucheble

Desuuuu, 2020-09-19:
I'm going to need some more details to be able to track down your issue. Please let me know:

* Precisely which button(s) are not working.
* On which screen this is happening.
* How the buttons behave. Do they make a sound?
* If there's a message showing up.
* What printer you have and what firmware/configuration you used.

ffnull, 2020-09-19:
> I'm going to need some more details to be able to track down your issue. Please let me know:
>
> * Precisely which button(s) are not working.
> * On which screen this is happening.
> * How the buttons [...]

On main screen. Button on center. For manual axes mooving. Im using your latest DGUS-Klipper reloaded for controller and lcd. Do all by your manual. Button doesnt recognize any touch. Im dont have buzzer, so no sound. Im do remake of new Creality Cr-5 PRO

Desuuuu, 2020-09-19:
Could you try updating to the [`dev`](https://github.com/Desuuuu/klipper/tree/dev) branch of my Klipper fork?

It has some changes that might fix your issue. You do not need to update the MCU firmware, just the host software (klippy).

ffnull, 2020-09-19:
> Could you try updating to the [`dev`](https://github.com/Desuuuu/klipper/tree/dev) branch of my Klipper fork?
> 
> It has some changes that might fix your issue. You do not need to update the MCU firmware, just the host software (klippy).

can you make simply step by step how to update?

Desuuuu, 2020-09-19:
If you cloned the repository into `~/klipper` like in Klipper's installation instructions, these commands should do:

```bash
cd ~/klipper
git pull
git checkout dev
sudo service klipper restart
```

ffnull, 2020-09-19:
Make your commands. Still doesnt work. Im think problem in DGUS firmware

ffnull, 2020-09-19:
On main screen, when im press MOVE icon - i dont move on next scree like pressing Themperature screen or others

Desuuuu, 2020-09-19:
I'm running the firmware without any issue (using the `dev` branch). I'm fairly certain this isn't an issue with the touchscreen firmware.

Please upload your `klippy.log` file, it might contain useful information.

ffnull, 2020-09-19:
interesting thing. when start print - print hungs while executing gcode DGUS_PRINT_START

ffnull, 2020-09-19:
[klippy.log](https://github.com/Desuuuu/DGUS-reloaded-Klipper/files/5249882/klippy.log)
add log

ffnull, 2020-09-19:
yay! its hangs and dont disable heaters. its dangerous!