Desuuuu/DGUS-reloaded-Klipper #2: [FR] Fysetc S6 v2.0 + Fysetc LCD

ETE-Design, 2020-12-30:
World be great if you also could make it work with FYSETC S6 v2.0 and their LCD 2.0 - https://github.com/FYSETC/FYSTLCD-2.0
I'm not into programming but have both parts so could help you test it if your up for the task :-)

Desuuuu, 2020-12-30:
If the touchscreen models they use are T5UID1 (I think they might be, but I haven't looked that much into it), the firmware should already be compatible.

In the case they're using a different model, I'm afraid I don't have the motivation and time to invest into porting over the screen firmware myself.

ETE-Design, 2021-01-01:
@Desuuuu Okey, do you have any screen shot's of the menu as it look for now?

Desuuuu, 2021-01-01:
> do you have any screen shot's of the menu as it look for now?

The screens are mostly images, which you can see [here](https://github.com/Desuuuu/DGUS-reloaded-Klipper/tree/master/project/DWIN_SET).

> is there an "Easy" way to add mixing extruder to the Touch Firmware?

Pretty much anything can be done but I wouldn't say there's an easy way. This would require modifications for both the host and screen firmware.

> Klipper Firmware for mainboard, is it possible to parse your changes to Kevin's Firmware, cause would like to always have the newest version possible, so it it many files you had made changes to?

You can do that by forking [my repository](https://github.com/Desuuuu/klipper) and then merging changes from [upstream](https://github.com/KevinOConnor/klipper) into your fork.

You shouldn't run into too many merge conflicts since I made very few modifications to the original files but there might be breaking changes that would require modifications to my code.

I'll see if I can setup a workflow to automate this process directly on my fork in the future.

closed 2021-01-13 by Desuuuu

Desuuuu, 2021-01-25:
I just setup a workflow to automatically merge upstream changes into the `snapshot` branch of my repository once a week.

There's no testing done on these versions though, so you might run into bugs due to incompatibilities with my code. Use at your own risk.