Desuuuu/DGUS-reloaded-Klipper #3: Question on setting up MCU connection to screen [cr10s Pro]

KillSwitch422, 2021-01-07:


Desuuuu, 2021-01-07:
IIRC the default serial port for the screen is indeed 2. Anyway, if you get past the boot screen, the serial connection is definitely working.

> There is no print status(not even temp or speed are shown) when started out of octoprint,

As specified in the [README](https://github.com/Desuuuu/DGUS-reloaded-Klipper), the print status relies on two specific commands (`DGUS_PRINT_START` and `DGUS_PRINT_END`) to work. You need to add them to your start/end scripts (or wherever appropriate).

> When idle, if I select the Move button it seems to launch the temp adjustment instead.

Sounds like the screen firmware wasn't flashed properly. Please follow [these steps](https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Flashing-the-firmware) carefully.

KillSwitch422, 2021-01-08:
>
> IIRC the default serial port for the screen is indeed 2. Anyway, if you get past the boot screen, the serial connection is definitely working. 

**Other option used during menucongif result in errors for reserved pins so I assume 2 is correct too**

Just re-flashed again, all seems successful, I can't get past the boot screen, without issuing a FW restart command in octoprint.  Once this is done connection seems fine.
> 
> Sounds like the screen firmware wasn't flashed properly. Please follow [these steps](https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Flashing-the-firmware) carefully.

**Operator error**, Apologies I did not use the archive, but the GitHub folder which does not have the required bin files.
Screen responding now. 

I will do some tests further as the abort button on screen kills the print, but then gets stuck (stays) on the printing screen, I assume it is because the `DGUS_PRINT_END`  is not run.

**Thank you for awesome work on these variants of printer. I used your marlin version before as well and it is so much better than other options.**

closed 2021-01-29 by Desuuuu