Desuuuu/DGUS-reloaded-Klipper #5: Z-offset not saved

krzim, 2021-02-21:
Perhaps I'm not doing something correctly but I set the z-offset using the touchscreen, home the z axis, and then back out to select `Save Config`. When I go back to check the z-offset, it has not remembered the setting. Should the z-offset be getting saved to the config by the touchscreen?

I'm running this on an Ender 5+ for what it is worth.

Desuuuu, 2021-02-21:
The offset adjustment is done using the `SET_GCODE_OFFSET` command, which sets a temporary offset.

For a permanent solution, you should adjust the offset directly in your configuration.

krzim, 2021-02-21:
That explains why it wasn't saving. I assumed it was doing a similar process to `PROBE_CALIBRATE`. Thanks for the quick answer.

closed 2021-02-21 by krzim

Desuuuu, 2021-02-21:
It would definitely make more sense to use the `PROBE_CALIBRATE` procedure on the leveling screen. I'll see if I can implement this.

krzim, 2021-02-21:
I was poking around the DGUS tool and project trying to figure out how to do just that. I'm sure you would be quicker at it than I would. I've spent the last few hours trying to figure out how the touchscreen communicates with the MCU and Klipper exactly. What relevant sections of the klipper codebase should I be looking at here?

Desuuuu, 2021-02-21:
The bulk of my work is in the [klippy/extras/t5uid1](https://github.com/Desuuuu/klipper/tree/master/klippy/extras/t5uid1) folder.

Most of the touchscreen logic is defined in the cfg files, while the Python stuff handles more generic stuff (setup, configuration parsing/execution, communication, interacting with Klipper, ...).

krzim, 2021-03-04:
I started looking into this but then my SKR Mini E3 v2 arrived and I pivoted to making that work. It looks like the combo of the DGUS LCD, Klipper, and SKR Mini E3 v2 isn't possible ATM. I've loaded vanilla Klipper on it in order to ensure things work well and then I plan to work on getting the stock touchscreen to work with the new board and Klipper. 

That may be a while considering I don't have a lot of experience with the codebase. Is there a particular reason you only made the AVR/LPC176X work with the DGUS screen?

Desuuuu, 2021-03-04:
It's most likely because I didn't have other boards to do testing. I can't remember any more specific reason.

Anyway, if you don't mind doing some testing, I can work on an STM32 implementation.

krzim, 2021-03-05:
I suppose this chain is a bit off topic for the issue (is there a better place for continuing this?) but I did get the new firmware flashed and can confirm that everything runs. The only issue I'm having is that the screen sticks on the boot image. I currently have jumpers connected between the board and the screen since the stock cable and connector do not work with this board.

I can verify that the `tx` pin is being held high with short bouts of it flickering low which I would expect from TTL serial. I'm a bit short on time today but I'll look into it tomorrow or this weekend and see if I can troubleshoot a bit more. 

Thanks for the rapid turnaround on this. Now that I see the additions you made in the commit, it makes sense. I was staring at the `Kconfig`, `serial_t5uid1.c`, and `Makefile` for the AVR boards trying to figure out exactly what I needed to implement for the stm32 boards. I knew those were key but didn't know how they linked to the hardware. Comparing the `serial_t5uid1.c` file against the board's pinout, I can see how it works.

Desuuuu, 2021-03-05:
Let's continue this discussion [here](https://github.com/Desuuuu/klipper/pull/21).

It would be great if you could give me more details like:
* How you configured klipper
* How you wired the touchscreen to the board