Desuuuu/DGUS-reloaded-Klipper #6: Print progress screen remains after cancelling print in Octoprint

krzim, 2021-03-06:
If I cancel a print in Octoprint, the print progress screen remains on the display and cannot be closed via any on screen options. The only way to get back to the main menu is to do a `restart` command in the Octoprint terminal.

I am running your fork of klipper on the STM32 MCU but I vaguely recall it happening on the stock AVR board as well. In both instances I am using the M73 plugin in Octoprint to send updates to the display.

I can provide whatever info you need to replicate.

Desuuuu, 2021-03-06:
https://github.com/Desuuuu/DGUS-reloaded-Klipper#prerequisites

https://github.com/Desuuuu/klipper/issues/9

closed 2021-03-06 by Desuuuu

krzim, 2021-03-06:
Ah yes I did set up my start/stop gcode in my slicer but didn't think to check the klipper issues for a solution to this. Apologies.