Desuuuu/DGUS-reloaded-Klipper #7: 2.4 TFT from Longer LK4 / XPT2046

totynho, 2021-03-26:
Hello, i have a longer lk4 with a STM32 MCU. The touchscreen is SID240320-8PCB-D board with XPT2046. If you search XPT2046 in google, you can see this being used directly in raspberry pi, and have many code avaliable. I don't know how hard is to port, but this model aparents to have many documentation in web... Marlin 2.0 with SID240320-8PCB-D configuration is easily found in the web too, i don't know if is possiblle transpile configurations to klipper / DGUS way... Can you look? I'm from Brazil, sorry about my bad english.

totynho, 2021-03-26:
[image]
[image]

totynho, 2021-03-26:
This lcd is used in many models of Longer, Alfawise and Digoo (i don't know if it is digoo)

totynho, 2021-03-26:
https://github.com/taburyak/STM32-touchscreen-XPT2046-HAL-SPI
https://github.com/topics/xpt2046

i found this, but I really don't know if is useful

Desuuuu, 2021-03-26:
Sorry but this is completely outside of the scope of this project, which is to add support for DGUS touchscreens.

closed 2021-03-26 by Desuuuu