Desuuuu/DGUS-reloaded-Klipper #8: automatic leveling does not work.

nxvash, 2021-04-10:
CR-10 Max, fresh install of your custom klipper which went very smoothly.  Flashed your LCD firmware and everything I have tested so far works great.  When I hit the Automatic button on the screen nothing happens.  I watch the terminal in OctoPrint and nothing shows up in there when you press the button either.  Any idea what needs changed to make this work?  Its not a huge deal as I can always run G29 from the terminal and save it with a M500, but it would be nice to have that functionality without having to have a laptop near by as my printers are in a different area of the house. 

Thanks for everything you did making this all so easy to install and get up and running. This is my first experience with Klipper and was extremely nervous about going that route but your setup made it a walk in the park.

Desuuuu, 2021-04-11:
Please provide your `printer.cfg`.

nxvash, 2021-04-11:
Sorry about that, I didn't think to attach that.  
[printer.cfg.txt](https://github.com/Desuuuu/DGUS-reloaded-Klipper/files/6290963/printer.cfg.txt)

Desuuuu, 2021-04-11:
That's happening because the screen only supports 5x5 meshes.

I'm working on a more customizable firmware that shouldn't have such a limitation.

closed 2021-04-11 by Desuuuu