Desuuuu/DGUS-reloaded-Klipper #10: Movement Page & Screen Alignment

Deneteus, 2021-04-24:
I have been playing around with the LCD menus and I cannot get to the movement section because it switches to fan speed info.
Is there some way to calibrate the screen in this release? I have experience using the Wanhao i3 Plus DGUS screen before this and it seems like the movement and temperature menus alignment are off. I initially thought the Z Offset screen was broken since it didn't save after being entered as well but I read the other thread about it not saving the info. The screen should probably say that the changes need to be added via the Klipper config.

Desuuuu, 2021-04-25:
Please follow [these steps](https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Flashing-the-firmware), including the optional flash reset one.

Deneteus, 2021-04-25:
> Please follow [these steps](https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Flashing-the-firmware), including the optional flash reset one.

I had actually already done that twice. I had the Tiny Machines firmware before I tested this.

Desuuuu, 2021-04-25:
Your description clearly indicates an issue related to the flashing process.

Something that could work is renaming the `13_touch.bin` and `14_variable.bin` files to the name they had in your previous firmware.

Deneteus, 2021-04-25:
Considering I flashed it 4 times with from the stock to the latest version of Creality firmware, from that to the Tiny Machines Firmware where I used it for a week and now to this. When I saw that the firmware had a mix of both was when I wiped it out using the method you described because I read the documentation. So it wiped the stock firmware, and the Tiny machines firmware out at that point. I wiped the memory card with the files out as well to make sure nothing was left over.

Desuuuu, 2021-04-25:
It does sound like the same bug described in this [issue](https://github.com/Desuuuu/klipper/issues/33).

There appears to be something weird going on with the flashing process, and renaming the files before flashing might actually help.

closed 2021-05-07 by Desuuuu