Desuuuu/DGUS-reloaded-Klipper #11: Can't display Printing on real print

Witawat, 2021-04-26:
i use version 1.0.2 but can't display Printing on real print
this printer show page home only

need setup or config display on printing mode ?

thankyou

Desuuuu, 2021-04-26:
https://github.com/Desuuuu/DGUS-reloaded-Klipper/wiki/Print-status

Whether you use OctoPrint or not does not matter. The print status display is controlled by Gcode.

closed 2021-04-26 by Desuuuu

KleBoR, 2021-04-26:
hi, i didnt wanted to open another "print time not displayed" thread cause mine is kinda working. the only thing im wondering is, that i have to insert the code into cura. if i use the octoprint start gcode it wount switch. anyone facing this also? 
another thing - is it possible to insert the eta instead of the time it just spent?

thank you very much